NorthernWidget/ALog-BottleLogger #8: I2C level shifting

awickert, 2017-04-01:
I2C level shifting was an issue; our plan (which we have completed -- this bug report is a note) is to restore the MOSFET-based level shifter and to connect the 3V3(!!!) pins off-board.

I have done this.
Not closing this yet because:
* @superchap123  hasn't yet reviewed
* We need to check into rumor that gate-source threshold must be > source - drain difference (don't think so from our tests, but want to double-check)
* We may want to add a (3V3) to the Silk layer on SDA and SCL.

awickert, 2017-04-01:
OK, Silk note added. Just review by @superchap123  needed.

superchap123, 2017-04-04:
@awickert I am reviewing the SMD branch file and the MOSFETS have not been restored.  Do you need to push a local copy yet or does this need to get fixed?

awickert, 2017-04-11:
@superchap123 We figured this out! And ordered the replacement boards (thanks!).

closed 2017-04-11 by awickert